tmolteno/TART #9: Image time is incorrect

tmolteno, 2017-07-09:
The image time appears to be 12 hours early. At 09:14 on Monday July 10 NZST it displays.

Recorded: Sun Jul 09 2017 21:14:59 GMT+1200 (NZST).

Correct behaviour is for the API to return UTC, and the browser to display in local time.

tmolteno, 2017-07-09:
The API http://tart2-raspberry/api/v1/imaging/timestamp uses isoformat() but does not include a time zone. Here is an example: 2017-07-09T21:25:03.794269. I suspect the error is in the JavaScript that should interpret this as an UTC time.

milletf1, 2017-07-09:
I can get it to display the time like 10/07/2017, 09:27:03.  Is that what you had in mind?

tmolteno, 2017-07-09:
I think that in web-app/src/app/services/imaging.service.ts, the getTimeStamp method should convert to local time, or set the time zone on a Date object to UTC. Then the javascript can display knowing the correct time zone.

tmolteno, 2017-07-09:
Alternatively the API should include a time zone of UTC in the string?

milletf1, 2017-07-09:
date.toLocaleString() outputs the time as I have displayed above.

tmolteno, 2017-07-09:
Perhaps the issue is browser dependent. I'm on a Chromebook, and the image from Monday morning shows a time 12 hours early (it does include NZDT, it just happens to be 12 hours out)

closed 2017-07-12 by milletf1