tmolteno/TART #23: API - Number of samples exponent in vis mode

maxscheel, 2017-07-26:
Number of samples exponent when changed, need to reconfigure the FPGA. Changing values breaks telescope process (API will still run, but telescope won't react)

psuggate, 2017-07-26:
The hardware won't automatically issue a reset & flush the buffers, if the exponent is changed mid-calculation, so you'll have to do that from the RPi, and change the web-app's state accordingly?

maxscheel, 2017-07-26:
Yip. Don't worry, I'll handle it. No action required from your side :)

tmolteno, 2026-10-02:
Reading the current code, a change to `/acquire/vis/num_samples_exp` only updates the config dict — the capture loop reads it once at `tart.start(...)` when the stream is set up, and the `vis_stream_reconfigure()` helper that would reprogram the FPGA is never called — so the fix is to reconfigure (or restart the vis stream) when the exponent changes; could someone confirm on a live telescope that this is still the behaviour before we touch the API?

---