tmolteno/TART #33: Extracting individual antenna signals

danieludick, 2018-03-16:
Good day, 

I would like to extract the individual (time sampled) signals from each of the antennas for our setup in SA. I am logging in to my Raspberry pi and I execute:

``
/git/TART/software/python/tart/tart/simulation $ python telescope_system.py
``

I am however getting the following error:

``
Traceback (most recent call last):
  File "telescope_system.py", line 13, in <module>
    rad = radio.Max2769B(sample_duration = 1e-5)
TypeError: __init__() got an unexpected keyword argument 'sample_duration'
``

The installation is also already up to date (I think). After pulling from the repo, I issued:

``
/home/pi/git/TART/software/python/tart_web_api and issue: sudo python setup.py install
``

Perhaps there is another setup.py to run?

danieludick, 2018-03-16:
There is another script that I located (test_radio.py), that looks useful here. When I run it however, then it also exhibits an issue:

``
pi@tart2-orange:~/git/TART/software/python/tart/tart/simulation/test $ python test_radio.py
Traceback (most recent call last):
  File "test_radio.py", line 10, in <module>
    from tart.simulation import spectrum
ImportError: cannot import name spectrum
``

maxscheel, 2018-03-16:
Hi Danie,
You went down the wrong rabbit hole and discovered lots of software/bit rot :1st_place_medal: 

Ideally you want to download the raw data file to your desktop. On your desktop should already have the tart python package installed (pip install tart)

I just added a simple script for time sampled signals from raw data pkl files in:
TART/doc/getting_started/time_sampled_signals.py

The latter is a stripped version of the spectrum generation from raw data which can be found here:
TART/doc/getting_further/spectrum/raw_data_info.py

tmolteno, 2026-10-02:
The extraction question was answered back in 2018 with `doc/getting_started/time_sampled_signals.py`, but the script you hit the error in is still broken on master — `telescope_system.py:13` passes `sample_duration=` to `Max2769B`, whose `__init__` takes `noise_level`/`n_samples` — so we will fix that call site and then close this.

---