tmolteno/TART #34: web-app file download path inconsistent when remote proxy adds extra path

maxscheel, 2018-03-16:
This is only an issue when wanting to download files externally. Internally (locally) routing is consistent.

When wanting to download raw data files from the browser (externally)
files are linked to:
https://tart.elec.ac.nz/raw/2018/2/1/23_36_51.819267_data.pkl
instead of:
https://tart.elec.ac.nz/signal/raw/2018/2/1/23_36_51.819267_data.pkl

tmolteno, 2019-07-11:
Leaving this issue open. Note that it only applies when downloading data from the public telescope that has been mapped to a different URL. Will fix by routing /raw and /vis to /signal/raw and /signal/vis on the public scope.

On direct connection to a TART via a browser, download works fine.

tmolteno, 2026-10-02:
The bug is still in `file-download.component.ts`: it composes download links from `window.location.host` only, so with the `/signal/` base href the app is built with, links point at `/raw/...` instead of `/signal/raw/...`; we will switch it to the document base href the API service already uses.

---