tmolteno/TART #37: Antenna positions not uploaded correcty to (SA) TART

danieludick, 2019-02-14:
As part of the S.A. TART calibration process, I measured the antenna positions. This was done according to the Jupyter notebook listed in https://github.com/tmolteno/TART/tree/master/doc/calibration/positions.

I  replaced all the measurements wiith our ones. Something that I did however change, was the antenna positions, which is entered into the following array (which accordingly is the distance from each antenna to the reference poles: x0, x1 and x2).

> m[0,:] = [1563, 855, 2618]

I assumed the above example was incorrect, as the order of the columns 1 and 2 is interchanged, i.e. it should be:

> m[0,:] = [855,1563, 2618]

After entering our data, I generated the attached JSON file: (note i merely renamed it to .txt for the upload to Github): [tart_positions_2019-02-14.txt](https://github.com/tmolteno/TART/files/2865369/tart_positions_2019-02-14.txt)

I then submitted the data to the TART, as follows:

> /usr/local/bin/tart_upload_antenna_positions --api http://146.232.222.105 --pw changethispw --file tart_positions_2019-02-14.json 

 and received the response:

> Response: {}

I then downloaded the antenna gains again from our TART (in order to verify that they are properly set), but they did not correspond to what I uploaded (see the attached file): [antenna_positions.txt](https://github.com/tmolteno/TART/files/2865378/antenna_positions.txt)

However, after putting our TART in Diagnose mode, the status map of the antennas changed completely:

[image]

This is not correct.

I am probably doing something wrong here. Should I also add the antenna positions as in the example, i.e. [distance from antenna to x1, ..x0, x2 ] ?

tmolteno, 2019-02-14:
Thanks for the report. Looking into it at the moment, and suspect that the API-uploaded telescope positions aren't sticking properly. Will update when I find more.

The positions can also be updated by modifying the json file in 
     software/telescope_web_api/config_data/calibrated_antenna_positions.json
and rebuilding the docker module (docker-compose build in the top level directory)
These positions are then permanently modified (the API uploaded ones don't survive a reboot)

tmolteno, 2019-02-14:
The notebook is very ambiguous, so I have updated it. The m[i,j] are the measurements (in mm) from x1, x0, x2 respectively - yes its very confusing!

tmolteno, 2019-02-15:
I have updated the tart-tools package. Please update with

    sudo pip install tart-tools --upgrade

This will report more useful error messages, as well as reporting 'SUCCESS' if the upload works. Also have updated the documentation to use the tart-tools binaries for uploading.

tmolteno, 2019-07-11:
Closing this as I believe this issue has been worked out.

closed 2019-07-11 by tmolteno