tmolteno/TART #41: TARTZA installation fails with docker-compose up

danieludick, 2019-06-14:
We are currently in a process of updating our TART in SA (i.e. TARTZA) to Python3. After installing Raspbian v9 (stretch) on a Pi 3B+, we still cannot get the new telescope software to run. 

We completed all steps detailed in the [README](https://github.com/tmolteno/TART/blob/master/software/README.md) but when we execute `docker-compose up` we get a range of errors, including the following:

> api_container_1   |   File "/usr/local/lib/python3.7/dist-packages/tart_hardware_interface-0.2.0b3-py3.7.egg/tart_hardware_interface/tartspi.py", line 8, in <module>
> api_container_1   |   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1145, in resource_filename
> api_container_1   |     self, resource_name
> api_container_1   |   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1715, in get_resource_filename
> api_container_1   |     return self._extract_resource(manager, zip_path)
> api_container_1   |   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1736, in _extract_resource
> api_container_1   |     timestamp, size = self._get_date_and_size(self.zipinfo[zip_path])
> api_container_1   | KeyError: 'tart_hardware_interface/permute.txt'

Do you have any recommendations for us?

danieludick, 2019-06-14:
Please note, we have not made any changes to the FPGA software (that is still running the original code that we received with our TART).

tmolteno, 2019-06-17:
I will look into this, and try and duplicate it here. Can you issue the following command on the pi (in the software directory)
 
    docker-compose logs > logs.txt

And then attach the file logs.txt (providing it isn't more than 1MB). Also the output of the following command:

    uname -a
    df -k

Which should get the exact version of the system and the amount of free space available.

danieludick, 2019-06-17:
Thanks @tmolteno . Attached is the [logs.txt](https://github.com/tmolteno/TART/files/3296170/logs.txt) file and here is the output from the other commands:

```
pi@tartZA:~/TART-master/software $ uname -a
Linux tartZA 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

pi@tartZA:~/TART-master/software $ df -kh
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  7.2G  6.7G  52% /
devtmpfs        460M     0  460M   0% /dev
tmpfs           464M     0  464M   0% /dev/shm
tmpfs           464M   30M  434M   7% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1   43M   22M   21M  51% /boot
tmpfs            93M     0   93M   0% /run/user/1000
```

tmolteno, 2019-06-18:
I think it is due to a mismatch in package versions. I am res-installing from scratch on a new machine. Should know by tomorrow morning (NZ time)

tmolteno, 2019-06-18:
I have reinstalled and it appears to be working. One issue might be the need to enable SPI on the raspberry Pi. I have made a few few updates to the README in TART/software (as docker-compose requires a few prerequisites to install).  I used a new RPi install disk and did...

1. Download new Stretch Lite installation from https://www.raspberrypi.org/downloads/raspbian/
2. Install onto new SD card (/dev/sdc on my system)
    unzip 2019-04-08-raspbian-stretch-lite.zip
    sudo dd if=2019-04-08-raspbian-stretch-lite.img of=/dev/sdc bs=4M
    sync
2. Insert uSD card into Pi and boot.
3. Log in an sudo raspbi-config
    - enable SSH server
    - enable SPI interface
    - take note of IP address
    sudo apt update
    sudo apt upgrade

4. Follow steps in software/README.md after logging in via SSH
    ssh pi@testpi.local
    
5. The Web interface appears and visibility mode can be enabled and data retrieved by SPI.

Could you check whether SPI was enabled on the Pi?

Tim

tmolteno, 2019-07-11:
Think this may have been due to line endings caused by transferring files from a windows host to the RPi SD card. Closing this issue for now. Please reopen if there are further problems.

19173296, 2021-04-13:
Hello. I am experiencing the exact same issue trying to update TART-ZA to the latest build. I followed this post in conjunction with the software README guide. I used a linux pc to create the Pi's sd card and copied the TART repo to the Pi. SPI and SSH is enabled and I can build and run the docker files, but It seems that the software_api_container keeps restarting every couple of seconds. The website is up, but I cannot log in with the default password. The latitude and  longitude stats also doesn't load. screenshot of the web interface:
[image]
 
'docker container ls -a' prints:
CONTAINER ID        IMAGE                     COMMAND                  CREATED             STATUS                          PORTS                  NAMES
b599dc30f638        software_web_app          "nginx -g 'daemon of…"   41 minutes ago      Up 38 minutes                   0.0.0.0:80->80/tcp     software_web_app_1
0befcfcb6bf1        software_api_doc_server   "nginx -g 'daemon of…"   41 minutes ago      Up 38 minutes                   0.0.0.0:8080->80/tcp   software_api_doc_server_1
4734f5353b27        software_api_container    "/bin/sh -c 'flask r…"   41 minutes ago      Restarting (1) 52 seconds ago                          software_api_container_1

'uname -a' prints: 
Linux tartza 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
and df -k prints:
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       15020844 3990816  10387076  28% /
devtmpfs          439916       0    439916   0% /dev
tmpfs             473196       0    473196   0% /dev/shm
tmpfs             473196   18268    454928   4% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             473196       0    473196   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   48782    209314  19% /boot
tmpfs              94636       0     94636   0% /run/user/1000

This is the docker logs:
[logs.txt](https://github.com/tmolteno/TART/files/6306119/logs.txt)

I can provide an direct ssh link to the pi if you want to have a look at the install on our pi
Thank you for your time

tmolteno, 2021-04-13:
OK. Looks like the permute.txt file in the tart_hardware_interface module is not being loaded. 

```
api_container_1   |   File "/usr/local/lib/python3.7/dist-packages/tart_hardware_interface-0.2.0b3-py3.7.egg/tart_hardware_interface/tartspi.py", line 8, in <module>
api_container_1   |   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1145, in resource_filename
api_container_1   |     self, resource_name
api_container_1   |   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1715, in get_resource_filename
api_container_1   |     return self._extract_resource(manager, zip_path)
api_container_1   |   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1736, in _extract_resource
api_container_1   |     timestamp, size = self._get_date_and_size(self.zipinfo[zip_path])
api_container_1   | KeyError: 'tart_hardware_interface/permute.txt'
```

This might be due to a change in how this is done. I'll look into it today.

tmolteno, 2021-04-13:
Can you check on the raspberry pi for the existence of a file. 

    docker-compose -f docker-compose-client.yml exec api_container /bin/bash
    find / -name 'permute.txt'

On our TART we get the following.
`root@0f3a0242593f:/app# find / -name 'permute.txt'
find: '/proc/19350/task/19350/net': Invalid argument
find: '/proc/19350/net': Invalid argument
/root/.cache/Python-Eggs/tart_hardware_interface-0.2.0b3-py3.7.egg-tmp/tart_hardware_interface/permute.txt
/python_code/tart_hardware_interface/tart_hardware_interface/permute.txt
/python_code/tart_hardware_interface/build/lib/tart_hardware_interface/permute.txt
`

19173296, 2021-04-14:
The container restarts before find can complete a search for the 'permute.txt' file. 
Results:
```
pi@tartza:~/software $ sudo docker-compose -f docker-compose.yml exec api_container /bin/bash
root@22a83b2fb234:/app# find / -name 'permute.txt'
root@22a83b2fb234:/app# pi@tartza:~/software $
```

19173296, 2021-04-14:
I solved the issue. The permute.txt was not in the tart_hardware_interface folder.
Copying the permute.txt from TART/hardware/FPGA/tart_spi/data to TART/software/python_modules/tart_hardware_interface/tart_hardware_interface/  before running docker-compose build did the trick.
Thank you for the help

tmolteno, 2021-04-14:
Excellent news. Glad I could help.

tmolteno, 2026-10-02:
This was fixed in 2021: commit `ecc051e` (merged in PR #60) ships `permute.txt` inside the `tart_hardware_interface` package and `setup.py` now declares it as `package_data`, which removes the `KeyError: 'tart_hardware_interface/permute.txt'` that was restarting the api container — ready to close, but please reopen if a fresh install still fails.

---

tmolteno, 2026-10-03:
Closing as verified: the fix is in `ecc051e` (2021-04-14, merged via PR #60) and is an ancestor of master.

Verification on current master: `hardware/FPGA/tart_spi/data/permute.txt` is tracked, **and** `software/python_modules/tart_hardware_interface/setup.py:19` ships it via `package_data={'tart_hardware_interface': ['permute.txt']}`, and `tartspi.py:8` loads it with `pkg_resources.resource_filename('tart_hardware_interface', 'permute.txt')` — so an installed package cannot hit the original missing-file error again. Please reopen if a fresh install still fails.

closed 2026-10-03 by tmolteno