tmolteno/TART #42: Use HealPix to generate web images

tmolteno, 2019-07-11:
There are at least two potentially useful libraries that port healpix to javascript.

- https://github.com/michitaro/healpix
- https://github.com/cds-astro/cds-healpix-rust

When the gui is redone, we should use these to image correctly.

tmolteno, 2021-11-10:
This is now underway on the next branch! We're using webassembly and rust as the language for imaging.

tmolteno, 2026-10-02:
Still valid and still not landed on master: the only healpix usage is `cdshealpix` in the rust front end on the `next` branch, while the Angular web app has no healpix at all, so we will leave this open until the `next` imaging rewrite makes it across.

---

maxscheel, 2026-10-03:
it's unclear to me what the intend is here? for which repo is this issue? The Vue frontend/three.js renderer is meant to be agnostic to the grid/shape of pixels on the sky - so it is left to the imaging method - currently gridless lib compiled to wasm - to provided the vertices and colour values.